UniExeterRSE/NetClamp #109: Clicking the Start button puts the camera into External Trigger mode

omarjamil, 2022-08-10:
To start the experiment, put the camera into exetrnal trigger mode. Clicking the `Start` button should put the camera into this mode. The button should then turn into a `Stop` button.

stephenpcook, 2022-11-01:
Work in conjunction with #115

omarjamil, 2022-11-06:
Current work in branch `start_stop_hookup` implements this. The camera is put in `external_trigger` mode and then put in `experiment_loop` where the ROIs are processed (should be one in #114 ) and the cell activity (0 or 1) is communicated to the arduino via serial port and a lit. If any active cells, the camera is triggered to capture an image. However, there are a couple of problems I am working through at the moment:

1. The arduino does not always trigger the camera. I need send empty lists to arduino to start with otherwise it does not process the active ones properly. 
 - Not sure what the solution is here yet
3. If arduino misses the camera trigger, the camera goes in infinite wait for images to be ready. The loop then gets stuck.
 - This could be solved by changing `timeout` value in `EventStream` in `dcam.py`. Need to test this.

closed 2022-11-10 by omarjamil